TR4W/TR4W #512: Show TR4W execution path

n4af, 2021-05-03:
It can be confusing knowing the location of TR4W presently executing.

n4af, 2021-05-03:
CTRL-P will now show TR4W_PATH_NAME
4.100.5

closed 2021-05-03 by n4af

ny4i, 2021-05-03:
Wasn’t CRTL-P some other command?

Just a recollection.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2021-05-03:
yeah, ptt. i gonna move ptt to ctrl-b

73, Howie

ny4i, 2021-05-03:
Just so I can share my thoughts, I am never thrilled when we change keystrokes. Users may get used to the prior one. Does it really need to have a shortcut? Can you not just have the menu option with no shortcut?

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2021-05-03:
Ok, i will change it back...

n4af, 2021-05-03:
show execution path is changed to CTRL-B