TR4W/TR4W #540: Change SHIFT KEY ENABLE

n4af, 2021-11-11:
Shift Key Enable previously bumped both the VFO and RIT.
This changes SHIFT KEY ENABLE to only bump the VFO frequency.

n4af, 2021-11-11:
4.105.6

ny4i, 2021-11-11:
I believe if RIT is on, it’s supposed to bump RIT. 

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

n4af, 2021-11-11:
KJ9C mentioned an arm injury and using shift to tune.
Actually that seems more useful- the way it is was set,
with SHIFT KEY ENABLE, the RIT AND VFO were bumped.

I commented out the RIT bump.

73, Howie

ny4i, 2021-11-11:
If RIT is off I agree. My point is if RIT is on, it should tune RIT. That is how it is documented I believe. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

closed 2022-01-16 by n4af