TR4W/TR4W #541: Investigate IF Shift on K3 and K4

ny4i, 2021-11-12:
We have reports that the K3 will shift the IF frequency on occasion. It is not clear what is causing this but this is a place to track research into the issue.

To see if the program is sending something strange, use the DEBUG LOG LEVEL CTRL-J config item. The code is already in MASTER to dump the data sent to the radio.Set it to DEBUG and you should get something like this in tr4w.log (in the program directory).

The log should produce something like this:

```
12 Nov 2021 14:28:21.655 19428 [13200] debug TR4WDebugLog  - Clearing main call window
12 Nov 2021 14:28:21.655 19428 [13200] debug TR4WDebugLog  - [TuneOnFreqFromCallWindow] Clearing Mults and QSO Needs Headers
12 Nov 2021 14:29:09.721 67494 [13200] debug TR4WDebugLog  - [SetRadioFreq] set frequency to 7025000
12 Nov 2021 14:29:09.721 67494 [13200] debug TR4WDebugLog  - FA00007025000;
12 Nov 2021 14:29:09.721 67494 [13200] debug TR4WDebugLog  - MD3;007025000;
12 Nov 2021 14:29:13.441 71214 [13200] debug TR4WDebugLog  - [ShowMultiplierStatus] MultNeedsHeader field set to 70
12 Nov 2021 14:29:13.441 71214 [13200] debug TR4WDebugLog  - [SetMultStatus] Entered with call = 70 and TempMult = 
12 Nov 2021 14:29:13.441 71214 [13200] debug TR4WDebugLog  - [SetMultStatus] Exiting early because NumberDifferentMults = 0
12 Nov 2021 14:29:14.129 71902 [13200] debug TR4WDebugLog  - [ShowMultiplierStatus] MultNeedsHeader field set to 703
12 Nov 2021 14:29:14.151 71924 [13200] debug TR4WDebugLog  - [SetMultStatus] Entered with call = 703 and TempMult = 
12 Nov 2021 14:29:14.151 71924 [13200] debug TR4WDebugLog  - [SetMultStatus] Exiting early because NumberDifferentMults = 0
12 Nov 2021 14:29:14.293 72066 [13200] debug TR4WDebugLog  - [ShowMultiplierStatus] MultNeedsHeader field set to 7030
12 Nov 2021 14:29:14.293 72066 [13200] debug TR4WDebugLog  - [SetMultStatus] Entered with call = 7030 and TempMult = 
12 Nov 2021 14:29:14.293 72066 [13200] debug TR4WDebugLog  - [SetMultStatus] Exiting early because NumberDifferentMults = 0
12 Nov 2021 14:29:14.621 72394 [13200] debug TR4WDebugLog  - [SetRadioFreq] set frequency to 7030000
12 Nov 2021 14:29:14.621 72394 [13200] debug TR4WDebugLog  - FA00007030000;
12 Nov 2021 14:29:14.621 72394 [13200] debug TR4WDebugLog  - MD3;007030000;
12 Nov 2021 14:29:14.621 72394 [13200] debug TR4WDebugLog  - Clearing main call window
12 Nov 2021 14:29:14.621 72394 [13200] debug TR4WDebugLog  - [TuneOnFreqFromCallWindow] Clearing Mults and QSO Needs Headers
12 Nov 2021 14:29:21.753 79526 [13200] debug TR4WDebugLog  - [SetRadioFreq] set frequency to 7040000
12 Nov 2021 14:29:21.754 79527 [13200] debug TR4WDebugLog  - FB00007040000;
12 Nov 2021 14:29:21.754 79527 [13200] debug TR4WDebugLog  - MD3;007040000;
12 Nov 2021 14:29:21.757 79530 [13200] debug TR4WDebugLog  - FR0;FT1;
```

ny4i, 2021-11-12:
The IS 9999; command does reset the IF to center on the K4. My K3 is not hooked backup yet.

n4af, 2021-11-12:
good enuff.
next question-- can DEBUG TRACE pick it up ??

Tks, Howie

ny4i, 2021-11-12:
Let me confirm the debug is right before we send to the serial port (and there is only one place we do). Otherwise, I will add some code to log there too.

ny4i, 2021-11-12:
Refresh my memory. Is this what should be in the CFG file to send this to the radio?

```
CQ CW MEMORY F12=<03>SRSIS 9999;<04>
CQ CW MEMORY F12 CAPTION=IF Cntr
```

n4af, 2021-11-12:
close but not quite. u need the srs'='

CQ CW MEMORY CONTROLF1=<03>*SRS=IS 9999;*<04>

ny4i, 2021-11-12:
Issue #543 logs this properly. I saw the IS 9999; Now all data we send to the radio is logged with TRACE.  Use TRACE instead of DEBUG.

ny4i, 2021-11-12:
BTW, I did not put the *SRS and it works OK.

The following works fine:
```
CQ CW MEMORY F11=<03>SRS=RC;<04>
CQ CW MEMORY F11 CAPTION=CLR RIT
CQ CW MEMORY F12=<03>SRS=IS 9999;<04>
CQ CW MEMORY F12 CAPTION=IF Cntr
```

n4af, 2021-11-12:
it seems to work.
u ready 4 me to ask Marv to run 4.105.8 TRACE ?

[LEFT] Calling WriteFile to write 3 bytes <IF;>
12 Nov 2021 17:17:18.258 19515 [9476] trace TR4WDebugLog  - Max wait time
(ms) -0
12 Nov 2021 17:17:18.258 19515 [9476] trace TR4WDebugLog  -
[ReadFromSerialPort] Read IF00007040140     -000000 0003000011 ;

ny4i, 2022-02-05:
I cannot recall if Marv still has the IF Shift issue?

n4af, 2022-02-13:
I think this can be closed.

closed 2022-10-15 by n4af