TR4W/TR4W #545: Move version info to its own file

ny4i, 2021-11-14:
Many times when I change config items like a new rig, I make changes to VC.pas. Many times the only change Howie makes each release to VC.pas is the version string. Those changes collide when I try to refresh my branch from Master. If the version info was in its own file, this would avoid collisions. 

I suggest the following constants be moved to their own file called version.pas.

```Delphi
const
 TR4W_CURRENTVERSION_NUMBER            = '4.105.7' ;  // N4af     New Release

  TR4W_CURRENTVERSION                   = 'TR4W v.' + TR4W_CURRENTVERSION_NUMBER; //  {$IF MMTTYMODE} + '_mmtty'{$IFEND};//{$IF LANG <> 'ENG'} + ' [' + LANG + ']'{$IFEND}{$IF MMTTYMODE} + '_mmtty'{$IFEND};
  TR4W_CURRENTVERSIONDATE               = 'November, 2021' ;

  TR4WSERVER_CURRENTVERSION             = '1.41';
```

n4af, 2021-11-14:
implemented 4.105.10

n4af, 2021-11-15:
If I move Version out-
unit version;

interface

implementation

const
 TR4W_CURRENTVERSION_NUMBER            = '4.105.10' ;  // N4af     New
Release

  TR4W_CURRENTVERSION                   = 'TR4W v.' +
TR4W_CURRENTVERSION_NUMBER; //  {$IF MMTTYMODE} + '_mmtty'{$IFEND};//{$IF
LANG <> 'ENG'} + ' [' + LANG + ']'{$IFEND}{$IF MMTTYMODE} +
'_mmtty'{$IFEND};
  TR4W_CURRENTVERSIONDATE               = 'November, 2021' ;

  TR4WSERVER_CURRENTVERSION             = '1.41';

end.
-----
and then set up Main.pas with Version in the USES

Main fails with a messages  like
[Error] MainUnit.pas(442): Undeclared identifier: 'TR4W_CURRENTVERSION'

n4af, 2021-11-15:
ok. I (think) it is good on github.

closed 2021-11-15 by n4af