TR4W/TR4W #553: CQWW fails to import zone

n4af, 2021-11-29:
Import of extant log fails to import the SRX_String zone content

n4af, 2021-11-29:
Assign 4.105.16

n4af, 2021-11-29:
fix 4.105.16

ny4i, 2021-11-29:
We need to enlist the user community and ask them to each pick a contest to test export and import to see if they run across any issues.

Tom

n4af, 2021-11-30:
Ok, UR7QM will investigate - if he is not a long hauler.
On the contest id in cbr, all seems ok in my submitted 4.105.16 log AND a new test log.

Howie

ny4i, 2021-11-30:
I mean contest ID in ADIF. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

n4af, 2021-11-30:
<sigh>.
yeah, i will take a look <ISSUE 554>

tks, howie

ny4i, 2021-11-30:
I looked but can’t figure out where the _ are being removed unless there is a function called doing that somewhere.

Tom

closed 2022-01-16 by n4af