TR4W/TR4W #556: QSO BEFORE MESSAGE is not sent in certain configuration

OH6QU, 2021-12-06:
If you set AUTO QSO NUMBER DECREMENT = TRUE and AUTO SEND CHARACTER COUNT = 4 then QSO BEFORE MESSAGE is not sent when working duplicate. TR4W sends the callsign and then stops and hangs PTT ON.

If you set AUTO QSO NUMBER DECREMENT = FALSE or AUTO SEND CHARACTER COUNT > 4 then QSO_B4 msg is sent correctly.

This is tested in TR4W v.4.106.0 with ARRL-10 contest as a non-NA station.

n4af, 2021-12-06:
I am not able to recreate this on 4.106.0
What do you have for AUTO DUPE ENABLE CQ ?

n4af, 2021-12-06:
just so i can get an idea of what is happening there:

ctrl-j and set your parameters for failure to get message
while in ctrl j also set DEBUG to TRACE

Do NOT save, just press escape (this to prevent extraneous future trace
entries)

now enter a duplicate call and produce the hang
then exit tr4w
send me TR4W.LOG in your main tr4w directory
also from main directory/SETTINGS, send me TR4W.INI

73, Howie

OH6QU, 2021-12-06:
I looks like the problem relates storing cfg & log files on the OneDrive. When storing files on the local hard drive this problem does not exist. I'll try to produce those files you were requesting.

ny4i, 2021-12-06:
Int is never recommended to store those files on OneDrive or Dropbox, etc—ESPECIALLY not the log file. I do set my backup log file to go to dropbox but that is it.

Tom NY4I

n4af, 2021-12-06:
ok. but one drive is not a tr4w problem.
what happens using 4.105.0 <https://tr4w.net/4.105/tr4w_setup_4_105.0.exe> ?

OH6QU, 2021-12-06:
OK, I'll be copying cfg and log files to the local hard drive when using TR4W and using OneDrive only as a storage for them.

OH6QU, 2021-12-06:
> ok. but one drive is not a tr4w problem. what happens using 4.105.0 <https://tr4w.net/4.105/tr4w_setup_4_105.0.exe> ?
> […](#)

I tested v.4.85.0 and v.4.99.0. Same problem exists with them as well. I haven't been using this combination of configuration setting for a long time and therefore not noticed this problem before. I have been using OneDrive for couple of years to run cfg and log files from.

n4af, 2021-12-06:
We, of course, want to address bugs in tr4w code.

However,  (my personal opinion) why I would *NEVER use QSO B4*:
(1) *Most of the time he copied my call wrong* and now has it right. QSO B4
just causes us to go around in circles losing time.
(2) *A bad spot/*has my call wrong. How long to debate with station vs
sending '599 #' ?
(3) He *did not originally come back to me*- I ERRONEOUSLY logged him first
time.
AHA, perfect reason to work him again.
#3 is why I praise stations who send MY call and shake my head at stations
who come back and send there own call, when
i responded correctly to theirs.

For some time, I ran NY4A M/M. I mades a point with our operators NOT to
argue about 'B4'--> send the exchange and move on.

My opinion/73, Howie

n4af, 2021-12-06:
Oh, boy, I left out a biggie why I WOULD NEVER send QSO B4.
There is an EXCELLENT chance that you have cost yourself the points of the qso !
IF NIL, OK but if no NIL then the points are lost.
It all comes down to lost time IMO.

ny4i, 2021-12-06:
I could not agree more. I would vote for working dupes to be the default setting in TR4W and no QSO B4 messages out into the default CW messages. Just work them. It never makes sense to argue. 

Maybe at one point dupes were penalized (but I cannot recall a time) but regardless, they are not now so we teach everyone at our Field Day for example to just work them.

Tom

n4af, 2021-12-06:
FB Tom -
I will ask on the forum and, if no strong objections, null QSOB4 out.

Tks, Howie

n4af, 2021-12-07:
4.106.2 removes AUTO DUPE ENABLE CQ option.
Force user to log all dupes.

OH6QU, 2021-12-07:
Removing AUTO DUPE ENABLED CQ makes sense as you described above. I keep on running TR4W cfg and log files from OneDrive as it seems to be working fine without need to send QSO MESSAGE BEFORE msg. That simplifies my life a great deal. Having cfg and log files in a single locations makes you confident you have the latest versions of them in use.

ny4i, 2021-12-07:
My caution about running the log in OneDrive is the program is not terribly efficient when it comes to reading the log. Rather than doing anything sophisticated like using slices of the log, it reads the entire log in far too often. This is a by-product of expecting the log would be local. So the configs on OneDrive would be fine but the log file in OneDrive may see excessive delays. If you have not run across this and it is not noticeable, that is fine. But as the log becomes bigger, it may be an issue. For small contests of a few hundred contacts, it may not be noticeable though.

Ton NY4I

closed 2022-01-16 by n4af