TR4W/TR4W #561: CQ160 fails to import QTH

n4af, 2022-01-16:
CQ160CW and CQ160SSB fail to import Domestic QTH.

n4af, 2022-01-16:
to be fixed 4.107.3

ny4i, 2022-01-16:
I’m curious. I’ve seen quite a few issues about importing contacts. Where are people importing them from? I would think if they are using TR4W as their logger, there would not be as much importing. 

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

closed 2022-01-16 by n4af

n4af, 2022-01-16:
I would hope it is not a common problem, but anytime the cbr is deleted and
a subsequent import is done to
recover, it would be germane.
For example if 2021 Sweepstakes has been submitted, the user might delete
the cbr since the adi can be imported.
Then a realization, before SS 2022, that the 2021 Initial Exchanges list
had not been exported and an import is run.

I suppose there is potentially also an issue with import from a brand x
software product ?