TR4W/TR4W #572: Add missing K4 Network commands

ny4i, 2022-02-07:
It looks like I missed a few commands in the network implementation.

Specifically, RITBumpUp, RITBumpDown, VFOBumpUp, VFOBumpDown.

closed 2022-02-08 by n4af

ny4i, 2022-02-10:
> Hola,
>
> is there a way to send the CW macros (F-keys) to the K4 internal keyer
> without CAT?
>
> 73,
> Martin, OK1RR
>
> Dne 07. 02. 22 v 18:41 Tom Schaefer napsal(a):

What do you mean without CAT? You have CWByCAT control with Ethernet or serial. When you sent that, the KY command is used to send the CW.

I may not be understanding your question though.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

ok1rr, 2022-02-10:
I meant without CAT via USB. I plan to use CWByCAT via Ethernet. No 
experience with this, I always used an external keyer for years. Now I 
would try the internal keyer and my idea was to evaluate the already 
present Ethernet connection. This stuff is very new to me.

73,
Martin, OK1RR

Dne 10. 02. 22 v 19:54 Tom Schaefer napsal(a):

ny4i, 2022-02-10:
Ok well if you set CWbyCAT, the radio will send CW no matter which interface you use. You can still use a USB connection on the K4 for CW via DTR (and FSK via RTS). 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

ok1rr, 2022-03-09:
Tested CW via TCPIP & K4

Although the radio control works perfectly with my K4, CW sending not. 
The spaces are unnaturally long, the words are torn and a question mark 
is added at the end of the message. Tested in both Linux (Wine) and 
Windows 7. Results the same. The program is therefore unusable.

73,
Martin, OK1RR
K4 #0300

Dne 10. 02. 22 v 23:57 Tom Schaefer napsal(a):

ok1rr, 2022-03-09:
I meant spaces between characters. The spaces between dots and dashes 
are OK, the characters sounding great...

Tested CW via TCPIP & K4

Although the radio control works perfectly with my K4, CW sending not. 
The spaces are unnaturally long, the words are torn and a question mark 
is added at the end of the message. Tested in both Linux (Wine) and 
Windows 7. Results the same. The program is therefore unusable.

73,
Martin, OK1RR
K4 #0300

Dne 10. 02. 22 v 23:57 Tom Schaefer napsal(a):

ny4i, 2022-03-09:
Let me check on it. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

ny4i, 2022-03-09:
Martin,

I verified this just now and I do apologize. It was a bug I discovered but forgot to fix.

I will get it done by this weekend.

Pleases use the WinKeyer or the just the RTS sending on the USB serial port in the meantime (that works well on the K4).

Tom NY4I 
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

ny4i, 2022-03-09:
> Begin forwarded message:

Adding Martin’s email.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

commit 13e357d8 (2022-02-08): Implement RIT UP/Down and VFOBump Up and Down

Close #572