TR4W/TR4W #583: On at least the K4, when in Data mode, the display in TR4W states NONDIGI.

ny4i, 2022-02-22:
I think I broke something. The radio states NONDIGI in DATA mode, NONCW in CW mode and NONSSB in SSB.

ny4i, 2022-02-22:
The issue appears the BAND is not being set. Might be an issue calculating the band from the frequency.

But if I set the band with Band UP, then it knows the band so it appears in the code that gets the IF response and sets the radio band.

n4af, 2022-02-22:
Tom - Would it make sense to label new branches as 'issue nnn' ?

ny4i, 2022-02-22:
I used to do that but sometimes if the changes were small, I would put multiple in one issue.

But this one is an example of I only have the one fix to get it int he March release so that branch will be Issue583.

BTW, we can also prune some of the really old branches. We have every branch ever created I believe.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

ny4i, 2022-02-23:
Well now I cannot duplicate this. I tried with the latest code in Delphi as well as the installed code in the latest install. Closing for now.

ok1rr, 2022-03-04:
Hi,

there is a bug - when a preprogrammed message contains > (RIT clear) 
program 4.109.0 crashes.

73,

Martin, OK1RR

ok1rr, 2022-03-05:
The last good one is 4.108.0.

The annoying bug appeared when the K4 was introduced.

73,
Martin, OK1RR

----------------------------

Hi,

there is a bug - when a preprogrammed message contains > (RIT clear) 
program 4.109.0 crashes.

73,

Martin, OK1RR

ny4i, 2022-03-05:
Are you connected via the network or serial cable?

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

ok1rr, 2022-03-05:
TR4W runs in Linux }no Windoze here).

Connections:

Network - while using TR4W. USB cable stays connected.

While using my logger (CQRlog) must use USB port (Hamlib 3.3) and K3 as 
radio type. Of course the network cable stays in its position.

73,
Martin, OK1RR

P.S. Still don't know how to use the internal keyer for macros with KY 
command. How should I configure the keying (CW/PTT part, see 
tr4w-ctrl.png) and/or Winkeyer }assume that it must be disabled).

Dne 05. 03. 22 v 14:30 Tom Schaefer napsal(a):