TR4W/TR4W #585: On a K4, when a preprogrammed message contains > (RIT clear), program crashes

ny4i, 2022-03-04:


n4af, 2022-03-04:
i wonder:
a) whether he meant on k4
b) whether also applies to xit clear:
procedure TK4Radio.RITClear(whichVFO: TVFO);
begin
   Self.SendToRadio(whichVFO,'RC','');
end;

procedure TK4Radio.XITClear(whichVFO: TVFO);
begin
   Self.SendToRadio(whichVFO,'RC','');
end;

ny4i, 2022-03-04:
Oops. That looks like a copy6 and paste error. I will fix that too.

I emailed to ask.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

ny4i, 2022-03-04:
My mistake. According tot he book, RC; clears both RIT and XIT. No way to clear just one.

n4af, 2022-03-04:
testing on K3, seems to work FB.

ny4i, 2022-03-04:
I tested with the clear command but not with the programmed message. It might not be calling same function. I’ll figure it out tonight. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

ny4i, 2022-03-05:
The code in RITClear received an Abstract run time exception. The code was calling the TNetRadio object incorrectly (earlier implementation model).

ny4i, 2022-03-10:
Reopening as the send CW is very choppy when sending to the network

PR #586 (2022-03-05): Issue585 RIT clear k4
Closes #585

This is ready to be merged.
PR #587 (2022-03-10): The code in SendCW was not processed when sending to a network
A common theme was to short-circuit the specific function routine like SendCW to just send the message to the Networked radio. SendCW had important code that needed to be run even in the network case. So I added network code to AddToOutputBuffer. This fixed the CW. Closes #585

Ready to be built so Martin can test. Thanks
commit 064c5f18 (2022-03-10): The code in SendCW was not processed when sending to a network

A common theme was to short-circuit the specific function routine like SendCW to just send the message to the Networked radio. SendCW had important code that needed to be run even in the network case. So I added network code to AddToOutputBuffer. This fixed the CW. Closes #585
closed 2022-03-10 by n4af