TR4W/TR4W #588: lang\tr4w_consts_ukr not in MASTER

ny4i, 2022-03-15:
I received a compile error that that {$IF LANG = 'UKR'}{$INCLUDE lang\tr4w_consts_ukr.pas}{$IFEND} could not be found.

This was the code I just synced from MASTER on GitHub. I check and file is not in lang but is the LANG set to UKR in the code committed to MASTER?

Looking in VC.pas, it says this:
const
//         LANG                                  = 'ENG';
//       LANG                                  = 'RUS';
//       LANG                                  = 'MNG';
//       LANG                                  = 'CZE';
//       LANG                                  = 'SER';
//       LANG                                  = 'ROM';
//       LANG                                  = 'GER';
        LANG                                  = 'UKR';
// not implemented

so I think you may have committed the UKR version.

Tom

n4af, 2022-03-15:
can you try again ?
master now has ukr.pas, but still reads 4.109.3.ukr

Can you work with the ukr master ?
I believe if you change vc.pas such that you
uncomment 'lang = eng' and comment 'lang = ukr'
it will compile normally ??

ny4i, 2022-03-15:
Howie - Regarding your point, "I believe if you change vc.pas such that you
uncomment 'lang = eng' and comment 'lang = ukr'
it will compile normally ??”

That is true. You normally leave the committed one in MASTER pointing to ENG. I think that was one issue plus the file was missing. I see the file there now so we should be good.

Thanks,

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

closed 2022-04-03 by n4af