TR4W/TR4W #592: Add an API to send Cabrillo to SCP collection  (presently K6TU)

n4af, 2022-04-03:
Watchful users of Super Check Partial will see a significant increase in the number of callsigns in MASTER and its derived files.  This month’s release (just posted live) has 40,940 calls in MASTER – up from 35,416 last month.

This step function in active calls is a result of a new API which allows logging programs to directly submit a log to Super Check Partial.  This removes the need for the contester to take the extra step of emailing the log to [logs@supercheckpartial.com](mailto:logs@supercheckpartial.com)<mailto:[logs@supercheckpartial.com](mailto:logs@supercheckpartial.com)>.  Logs are submitted securely to Super Check Partial over HTTPS and are “signed” by the logging program before submission to authenticate the origin of the log.

SCP API support has already been added to N1MM+, DXLog.net, SkookumLogger, N3FJP’s loggers and Writelog.  This support is in published releases for N1MM+ and DXLog.net, selected contests for N3FJP loggers  and in beta versions for SkookumLogger and Writelog.

The supporting logging programs ask the contester for their consent to upload the log to Super Check Partial – by design this is an opt-in process.  Logs are NOT submitted without the contester’s consent.

ny4i, 2022-04-03:
> This step function in active calls is a result of a new API which allows logging programs to directly submit a log to Super Check Partial. This removes the need for the contester to take the extra step of emailing the log to ***@***.******@***.******@***.***). Logs are submitted securely to Super Check Partial over HTTPS and are “signed” by the logging program before submission to authenticate the origin of the log.

The Indy classes should make this easy ( especially with the TLS https call).

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

ny4i, 2022-04-03:
I checked the SCP site but did not see the API. Where is the API to submit the logs documented?

I presume there is a URL to post the file.

Tom

n4af, 2022-04-03:
Good question.
I think Stu handles via direct request:
"
If you are the author of a logging program and would like to add the API support to your program, please contact me directly for documentation etc.
"

Possibly via stu@k6tu.net
his update on the above came from
stu@ridgelift.com

ny4i, 2022-04-04:
To avoid any issues with sharing the key, I suggest we create a unit with an object that can be called. It will then be included in the code as a library (dci) but we will keep the source code out of it. The source can be in a private repo.

So who is adding this? I can write the unit to do the log upload, then it can be implemented in the TR4W code.

ny4i, 2022-04-04:
I spend some time looking into this project. The hard part is going to be to find a Delphi7 implementation of the SHA256 hash function. I have some leads but most encryption libraries are setup for newer versions of Delphi. The one I tested was not compatible with D7 as it used later Delphi syntax. (class functions).

n4af, 2022-04-04:
argh. Well, it was worth exploring.
Hard to understand the need for encryption of cabrillo data submission ???

ny4i, 2022-04-04:
It’s hashing to not overwhelm the server. We don’t actually encrypt the log. 

Let me take another look. I might be making this too hard. 

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

ny4i, 2022-04-04:
Yes I made this too complicated. I can generate the hash outside of the program. 

Base64 I have which is used to upload the log. 

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

ny4i, 2022-04-05:
Well we did need to hash in the program. I found the routines in Indy to set it. I also found the code to set the TLS version to 1.2. This has a few moving pieces including items such as new files that need to be installed in the program directory. I will check those in when the time comes. There will also be a new DCU called uSuperCheckPartialFileUpload.dcu. The source code will NOT be in this repo. This is to protect the implementation details (per the request of SCP/K6TU).

We do need to come up with an interface to do this send. Perhaps something on the tools menu to to upload and also an option after the CAB generation to ask if they want to upload the file automatically?

n4af, 2022-04-05:
Hi Tom -

I would just go with an option to send the log to K6TU whenever   cabrillo
is generated.

Tks, Howie

ny4i, 2022-04-05:
OK. Maybe some sort of acknowledgement after the cab file is created? Or do we want to force it even if they do not send create a CAB? If so, when would we send it?

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2022-04-05:
I suppose it could either be a query every time cab is generated -or- a
ctl-j option to generate scp to k6tu upon cab generation (my pref-but...)
perhaps that begs the question (on the later option)-
How does Stu handle multiple cabs/same contest/same submitter ? If he
throws away earlier TX then it is moot.

Tks, Howie

ny4i, 2022-04-07:
I committed a branch called Issue592. Please download that branch and see if this works for you.

I added two required DLLs to the target folder so we need to make sure they end up in the same directory as the EXE file when built. If it makes more sense to add these to the INSTALL script, that is fine. Maybe you can copy them from he target directory so I can update they int he repo when needed.

Also, if you do actually send the file, please only send real contest logs. Stu asked me to send real contacts and not test ones. I only have the production URL set and not the test URL set for development testing.

One last item, I'm not thrilled with the way the Export Cabrillo dialog works. It seems clumsy. We click ok and the edit window pops up with the Cabrillo file. Only when I close that edit window, do I then ask if the user wants to send the file. Then you are left with a Generate Cab dialog that you have to CANCEL. Maybe that its done in case you want to edit the form but the CANCEL is a bit clumsy. I am not sure of a good alternative.

n4af, 2022-04-09:
[fix 4.110.1](https://tr4w.net/4.110/tr4w_setup_4_110.1.exe)

ny4i, 2022-04-09:
I’m still working on this. I added a confirmation dialog after the upload. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

n4af, 2022-04-09:
ok. let me know when it is qrv to be packaged...

n4af, 2022-04-09:
i am bothered that there is not UI f/b on the process:
if I take down internet and go thru the process, it acts the same as if
internet was up ??
No indication that the TX to SCP failed ?

ny4i, 2022-04-09:
Give the pull request I Just did a try. See how that works for you.

I just used a string. I need to change to a TC_ constant so we can get the translation too.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

closed 2022-04-09 by n4af

ny4i, 2022-10-11:
What I’m fixing is the message box showed up behind the create Cabrillo dialog. 

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

ny4i, 2022-10-11:
Yes I fixed that but still testing. 

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

PR #593 (2022-04-09): Added confirmation dialog after upload or on error
Fixed #592 

Added Yes/No after the Cabrillo preview window is closed to ask if the user wants to upload the log file.