TR4W/TR4W #599: ARRL-DIGI missing Grid header

n4af, 2022-05-15:
There is a display problem in cabrillo:

START-OF-LOG: 3.0
CREATED-BY: TR4W v.4.111.1
CALLSIGN: UR7QM
CONTEST: ARRL-DIGI

QSO: 14000 DG 2022-05-15 0516 UR7QM   KN76\Ѓ     HA1RB          JN8

There is no such problem in cabrillo formed in WSJT-X:

START-OF-LOG:3.0
LOCATION: KN76QT
CONTEST: ARRL DIGI
CALLSIGN: UR7QM

QSO: 14075 DG 2022-05-15 0516 UR7QM        KN76          HA1RB        JN86

ny4i, 2022-05-15:
I saw that once but could not reproduce it with the debugger. I’ll check again. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

n4af, 2022-05-15:
I have written to ARRL to see what it should be.
I could not find a CBR header example for this contest.

ny4i, 2022-05-15:
I think I asked when I created it but using this as a guide, https://wwrof.org/cabrillo/cabrillo-v3-header/ , ARRL-DIGI is right. 

Also, WSJT-X is using location wrong. It should be GRID-LOCATOR: text

Locator is for section or other info. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

ny4i, 2022-05-15:
I can fix these today. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

ny4i, 2022-05-15:
I added GRID-LOCATOR plus I found an issue in the MYExchange and HIS Exchange section. There was an array used for CABRILLO_MYEX that really should have been zeroed out before each contact is written. I have seen random errors where there will be some garbage characters after the exchange. This all depended what was in memory which explains why I would see it on and off.

closed 2022-05-15 by n4af

n4af, 2022-05-16:
The ARRL Log submission page uses some of the information provided in the upload questionnaire to determine entry category, which is independent of the information in the Cabrillo header.  Categories for the most part are selected on the uploader webpage.

The CONTEST: header as ARRL-DIGI is correct.

LOCATION: should be the ARRL/RAC Section (or DX for Non US/Canadian stations).  If a grid is in this field upon upload, it is replaced with the section or DX that is provided by the participant in the upload questionnaire page.

However, If a grid location is to be entered the header, it should be in the GRID-LOCATOR: field, not LOCATION: This often

The station’s sent 6 digit grid should be included in each QSO line, which would be used for scoring:

QSO: 50 DG 2022-06-07 1818 N7KSI CN86MN KD7UO CN87XJ

If you have any other questions, please feel free to contact me.

73,

Paul Bourque, N1SFE

Contest Program Manager

ny4i, 2022-05-16:
I am going to email Paul if you are ok. I do not believe we get the 6 digit grid from WSJT-X so I do not know how we could ever log that.

Tom

n4af, 2022-05-16:
RR73,hSent from my U.S. Cellular® Smartphone

"target": "https://github.com/n4af/TR4W/issues/599#issuecomment-1127734530",

"url": "https://github.com/n4af/TR4W/issues/599#issuecomment-1127734530",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"name": "GitHub",

"url": "https://github.com"

}

}

]

ny4i, 2022-05-16:
> This is lifted directly from the rules so my guess is Paul was operating from memory.
>
> Exchange: Send 4-digit Maidenhead grid-square locator (www.arrl.org/grid-squares <http://www.arrl.org/grid-squares>).
>
> This is a typical FT8 contest contact:
>
> CQing Station: CQ TEST N1SFE FN31
>
> Answering Station: N1SFE W9SZ EN50
>
> CQer: W9SZ N1SFE R FN31
>
> Answer: N1SFE W9SZ RR73
> CQer: W9SZ N1SFE 73 (or any appropriate acknowledge message)

4 digit grids it is.

From Paul N1SFE, 

Good morning,
 
You are correct, the exchange for the contest is a FOUR digit grid.  I was thinking about the other distance scored events ARRL has, where we use 6 digit grids.  My example was taken from a 222 and Up log, which uses 6 digit grids.  In the ARRL Log uploader page, we request a 6 digit grid for the club competition scoring, but in the log entries themselves, we require four digit grids in the Digital Contest.
 
Let me know if there’s anything else you need clarification on.
 
73,
Paul Bourque, N1SFE
Contest Program Manager

PR #600 (2022-05-15): Added grid-locator to header
Added GRID-LOCATOR to the header as that is a valid field. This closes #599.

Also, cleared the Cabrillo Exchange fields before each contact is written. There was random data appearing in the CAB file.

Also, fixed an outstanding issue in the same place where the operators array sorted from the log is placed in the order of the number of contacts made. This seems more based on merit. The only place this really applies is for a multi-op and when a certificate is received. The calls are typically listed in the same order as the Operators field. Minor issue.
commit a23760db (2022-05-15): Added grid-locator to header

Added GRID-LOCATOR to the header as that is a valid field. This closes #599.

Also, cleared the Cabrillo Exchange fields before each contact is written. There was random data appearing in the CAB file.

Also, fixed an outstanding issue in the same place where the operators array sorted from the log is placed in the order of the number of contacts made. This seems more based on merit. The only place this really applies is for a multi-op and when a certificate is received. The calls are typically listed in the same order as the Operators field. Minor issue.