TR4W/TR4W #605: version checked into MASTER has a version of 4.111.3.

ny4i, 2022-06-05:
I am not certain but it looks like the version in MASTER is 4.111.3. I believe the current version is 4.112? Was this not committed to MASTER? I guess I am still not certain how this is built as in my view the steps would be...

Commit everything to MASTER
Fetch MASTER
Build

I ask as when I went to edit5 the code to remove the STATE tag for a grid, it was not there so there is something out of step between what was built and what is committed into MASTER.

n4af, 2022-06-06:
Hi Tom -

Could you test 4.112.1 <https://tr4w.net/4.112/tr4w_setup_4_112.1.exe> to
ensure it picked up #605 ??

tks, h

ny4i, 2022-06-06:
The version you built previously and put not he website was showing 4.112 so that was not the issue. My question was that the version.pas committed into MASTER did not show 4.112 so it appeared that you were building things that were not in MASTER. That could lead to other issues where a patch is applied to MASTER via pull request but it does not make it to the machine from which you build. I checked the version.pas committed to MASTER and it looks good now.

The other curiosity was that the version you build for June has the bug with the STATE being added to ADIF for a GRID, but when I stepped through the version committed to MASTER, it did NOT do this. THat tells me there is a difference in other code too. That is the real issue at heart.

ny4i, 2022-06-06:
I installed 4.112.1 and the version is OK but it still emits <STATE:4>GRID. The version in MASTER does not so I guess I will research any changes in this area. Do you recall any?

n4af, 2022-06-06:
Hi Tom - I think a change I was doing for ne9u got into 4.111.3 and put our
two masters out of sync.
I think 112.0 is now ok, can you pull that in as master and redo 604 ??

ny4i, 2022-06-06:
The odd thing is the code in master does not have the error so I made no changes. But what you compiled this morning and the previous 4.112 did. I am at a loss to explain that.

Tom

n4af, 2022-06-06:
ok. ii will try to rebuild from 4.111.1 forward to 4.112.0 as master.

ny4i, 2022-06-06:
I think 4.111.3 had important fixes though

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

n4af, 2022-06-06:
as long as i can find the individual fixes for 4.111.3

ny4i, 2022-10-11:
> ok done.
> what is a good piece of code to verify 604 took ?
>
> git fetch
> remote: Enumerating objects: 33, done.
> remote: Counting objects: 100% (33/33), done.
> remote: Compressing objects: 100% (9/9), done.
> remote: Total 33 (delta 24), reused 32 (delta 24), pack-reused 0
> Unpacking objects: 100% (33/33), 12.58 KiB | 74.00 KiB/s, done.
> From github.com:n4af/TR4W
> * [new branch] 4.111.2 -> origin/4.111.2
> * [new branch] 4.111.3 -> origin/4.111.3
> * [new branch] 4.112.1 -> origin/4.112.1
> * [new branch]
> 604-in-arrl-digital-contest-the-adif-file-exports-state-with-a-grid-square
> ->
> origin/604-in-arrl-digital-contest-the-adif-file-exports-state-with-a-grid-square
> * [new branch] Issue599 -> origin/Issue599
> fd33bfc..8bd6b59 master -> origin/master

I’ll have to check. I have a meeting so it will be a bit later. Thanks 

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

closed 2023-07-27 by ny4i