TR4W/TR4W #635: On ADIF export, if there is an different operator field, include STATION_CALLSIGN

ny4i, 2023-02-07:
LOTW needs to have a STATION_CALLSIGN field of the callsign used if there is an OPERATOR field. This would make it clearer too when exporting so add STATION_CALLSIGN to the ADIF export. I would just do it in all cases but I found this in WINTERFIELDDAY.

ok1rr, 2023-02-11:
There is no visible TCP/IP parameter settings in the corresponding 
window. The settings can be copied from previous versions (settings -> 
tr4w.ini), all works fine. The [Radio] section may look like this:

[Radio]
RADIO ONE TYPE=K4
RADIO ONE CONTROL PORT=TCP/IP
KEYER RADIO ONE OUTPUT PORT=NONE
RADIO ONE CAT RTS=OFF
RADIO ONE CAT DTR=OFF
RADIO ONE KEYER RTS=OFF
RADIO ONE KEYER DTR=CW
RADIO ONE BAUD RATE=38400
RADIO ONE NAME=K4
RADIO ONE IP ADDRESS=192.168.88.249
RADIO ONE TCP PORT=9200

An attempt to set up the parameters from the program results in 
situation as in actual-tcpip.jpg however it should be as in 
previous-tcpip.png.

73,
Martin, OK1RR

ny4i, 2023-02-11:
Sounds like a resource file issue on that build. What does this have to do with ADIF? OR did you hijack another issue to report this?TomPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771

ny4i, 2023-02-11:
Please open a new issue to track this one. TomPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771

ok1rr, 2023-02-11:
OMG...

This has nothing with Re: [n4af/TR4W] On ADIF export, if there is an 
different operator field, include STATION_CALLSIGN (Issue #635)

At least I have experience for next time. So I won't be reporting any 
more bugs...

Martin, OK1RR

Dne 11. 02. 23 v 13:41 Tom Schaefer napsal(a):

ny4i, 2023-02-11:
Just open another issue in GitHub. TomPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771

commit 9a4353cb (2023-03-04): Fixed WFD Import and SRS commands for Net

Added code to call the NetObject.SendRadio command if the SRS commands are used. Also added code to fix the importing of WinterFieldDay files. This closes #635 and closes #637 as well as the WFD Import which did not have an issue.
closed 2023-03-04 by n4af