TR4W/TR4W #639: Add function key commands for <MESSAGE1>, <MESSAGE2>...<MESSAGE8>

ny4i, 2023-03-06:
Rather than require the user to code a command sequence, add support for sending the values of CW and DVK memory 1 - 8. This will be different for each radio that supports it. I will start with Elecraft of course. Note the K4 commands are different than the K3.

ny4i, 2023-03-06:
The pattern for this would be just turning the RIT off so the radio objects should implement functions called PlayMessage(msgNum: integer)

ny4i, 2023-03-06:
We also need a command to say StopMessage to hit call when an ESC is hit. Note some radios may not support interrupting the message. Elecraft does.

ny4i, 2023-03-07:
Icom 7610 command is 28, no sub cmd, 00 to stop voice memory, 01 - 08 to play voice memory 01 - 08 respectively so 28  01 to play voice memory 1.

ny4i, 2023-03-07:
K3 is SWT01; to play memory 1 and SWT04 to play memory 04. K3 just has 4 memories.

ny4i, 2023-03-07:
Some Yaesu radios support Playback with the PB command. On the FT-710 for example, PB00 to stop, PB01 to play memory 1 and PB05 to playback memory 5 (it goes up to 5).

ny4i, 2023-03-14:
I have the basic support now in ProcessCommands. The commands ended up being PLAYMESSAGE_ACTIVE and PLAYMESSAGE_INACTIVE. I am not sure if it would ever be needed to play a voice message on the inactive radio but this seems more flexible. 

I still do not like that we have to embed a <03> CTRL-C and terminate with a <04> CTRL-D. That is really clunky. I guess I am not sure why we really need to have that anymore. I would still support it as I am sure there are plenty of macros that have it, but I do wonder if I could change the code to just look for the command if CTRL-C is not the first character of a command.

n4af, 2023-03-23:
Hi Tom -
I see  
* [new branch]      voice-msg-commands -> origin/**voice-msg-command**s
Is this ready to integrate ? It would be 4.120.4 ?

ny4i, 2023-03-27:
Kenwood commands as follows:

- PB0; to stop
- PB1; to play memory 1
- Same for 2 and 3
- PB4; to play memory 4

ny4i, 2023-03-29:
To play messages, instead of adding the actual CAT commands, you can include commands to PLAYMESSAGE and the program figures out what commands to send to the radio. For example, 

```
[Messages]
CQ SSB MEMORY F1=<03>PLAYMESSAGE_ACTIVE=1<04>
CQ SSB MEMORY F1 CAPTION=DVK M1
CQ SSB MEMORY F2=<03>PLAYMESSAGE_ACTIVE=2<04>
CQ SSB MEMORY F2 CAPTION=DVK M2
QSL SSB MESSAGE=<03>PLAYMESSAGE_ACTIVE=3<04>
```

You can of course still send native commands the regular way with the SRS commands but this saves you having to look this up. You can also set the commands in the O section of the P. The above example shows where I set the QSL SSB Message to play a memory. SO when I am calling CQ, this plays memory 3 after I hit ENTER to log the callsign.

I have not tried the INACTIVE memory as I do not have two radios setup. Also, this was only tested on a K4 using both TCP/IP and a serial connection. I have not checked it against Icom or Yaesu but the commands are from the manual. This has to be tested.

commit 43095391 (2023-03-29): Ready to merge into MASTER

Added Kenwood and Yaesu radio commands. Also added commands to stop playing the message when ESC is hit. This needs to be tested on radios other than the K4. This closes #640 and closes #639 .
PR #642 (2023-03-29): Ready to merge into MASTER
Added Kenwood and Yaesu radio commands. Also added commands to stop playing the message when ESC is hit. This needs to be tested on radios other than the K4. This closes #640 and closes #639 .
closed 2023-03-29 by n4af