TR4W/TR4W #656: Improve format of summary data

n4af, 2023-06-19:
4.123.6 attempts a better alignment of contest summary posted to 3830.

ny4i, 2024-06-25:
When I generated a summary for ARRL Field Day to put the data into 3830, the format of the SUMMARY report was are follows:

```
BAND   Raw QSOs   Valid QSOs   Points     Mults
 __________________________________________________________________
    80  CW        1             1        2         0
    40  CW       36            36       72         2
    40DIGI       27            27       54         3
    40 SSB       85            85       85         1
    20  CW       49            49       98         0
    20DIGI      110           107      214        16
    20 SSB      531           530      530        22
    15DIGI      231           214      428        27
    15 SSB       89            89       89         5
    10  CW       18            18       36         6
    10 SSB       41            40       40         1
     6DIGI       47            45       90         0
 __________________________________________________________________

 Total         1265          1241     1738        83

```
But the form on 3830 is as follows:

[image]

It would be easier if the summary was output in the same format. If there is a historical reason to keep the summary as is, then create a new report called 3830 Summary:
```
         CW     PH   Dig
160
 80
 40
 20
 15
 10
  6
------------------------
Total
```

ny4i, 2025-07-20:
Here is the score entry page from 3830. Our summary format requires some thinking to convert it to the 3830Scores format. As mentioned, it would nice tro have an easyt report that has the qso and multiplier grid inthe same format as 3830Scores.

[image]