TR4W/TR4W #688: Lock the bandmap window while updating to prevent flashing

ny4i, 2023-09-01:
During frequent updates, the bandmap flashes from updating so frequently. One possible solution is to lock the window with the Win32 LockWindowUpdate function. 

I do not recall if I tried the this before.

Using this ticket to track it.

https://learn.microsoft.com/en-us/windows/win32/api/winuser/nf-winuser-lockwindowupdate

ny4i, 2023-09-01:
The LockWindowUpdaes suggested to send the message [WM_SETREDRAW](https://learn.microsoft.com/en-us/windows/desktop/gdi/wm-setredraw) instead. I tried that and while it made some difference, the BM still flashes.

One thing I learned is it appears the telnet code only uses a thread for the connection. I thought I must be reading that wrong, but when I added a sleep (250) to sleep 250 milliseconds, the whole UI slowed down. My thought was to limit how often the update updates but the telnet thread should be in a thread instead. 

Another thought is to set a TTimer to fire every 250 milliseconds and do the update. 

Even without adding full threading for telnet (which really should happen), that can be tested by setting BMDELAY to 250.

n4af, 2023-09-01:
alas-
the spot must sync to the display. else, clicking on a b.m. entry displays
the wrong call entry/frequency..
likewise, delaying spot processing results in lost spots.

ny4i, 2023-09-01:
Normally, the way to handle that is to store an object with the list box entry which is the spot data. That way clicking on any spot will always give the correct info. I’m not sure how to do that yet with Win32 but I’ll look. TomPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771

ny4i, 2023-09-01:
The above is wrong. The Telnet code is definitely in a thread but the part where it calls DisplayBandMap is not so more checking is required.

ny4i, 2023-09-03:
I'm quite confused. I am in the state of How could this possibly work reliably? When a spot is double-clicked, I see the code get an index by sending the message, LB_GETITEMDATA. But I cannot find anything related to the spot that does the corresponding LB_SETITEMDATA. Typically you would set an index then store that with the listbox entry. When accessed, you get the index (integer) back and use it to index a list (FLIST in this case).  

The other thing that is missing considering this is threaded code is any sort of locking around the index creation. The pattern should be grab a lock (TCriticalSection), assign the add the spot to the SpotList object, add the value to the bandmap list box including the LB_SETITEMDATA call), then free the lock.  In threaded code (which Telnet is, this is hit or miss. 

I do see some code in uTelnet that uses LB_SETITEMDATA but that is for the entry in the telnet window. It might be possible that the whole telnet record is passed to the spotLIst but that seems odd.

Looking at this code (and it is definitely a big pile of spaghetti), I would say a completely new object should be rewritten to manage the spots. The SpotsList object is the right idea but this code is really rough.

I of course could just not be understanding it fully yet but I am not following parts of it.

A simpler thing might be to just put a criticalsection around where the spot is created in the SpotsList and anywhere the SpotsList is modified (Add, Delete, etc).

n4af, 2023-09-04:
Hi Tom -

it is triggered by Uspots proc TDXSpotslListSetCursor
 FCurrentCursorFreq := *GetBMSelItemData*;
Returns the entry # in the BM.

ny4i, 2023-09-04:
What I can’t find is the corresponding SetItemDataPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771

n4af, 2023-09-04:
The call stack looks like:
LBN_Selchange
SpotsList.SetCursor
GetBMSelItemData <<< line 536 uSpots
FCurrentCursorFreq :=

ny4i, 2023-09-04:
Yes but I’m order for the index to come from GetBMSelItemData, there has to be a corresponding SetItemData. I don’t see that. TomPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771

ny4i, 2023-09-04:
I will add sone debug messages around the AddSpot to track the index. TomPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771

n4af, 2023-09-19:
i can set bm to update every x filtered entries in the cluster window.
setting it to every other entry halves the blinking and updates by two entries.
Q: should this be a CTRL-J parm set by the user for 'update very x spots'  ?
Let me put out a '[every other spot'  update]'(https://www.tr4w.net/4.126/tr4w_setup_4_126.beta.exe) and you can see if that works.

n4af, 2023-09-20:
[TR4W beta2](https://www.tr4w.net/4.126/tr4w_setup_4_126.beta2.exe) should greatly reduce the update frequency. 
BM runs one spot behind the cluster window.

ny4i, 2023-09-20:
I recall you mentioned if you slowed it down, the BM item you click on might change to the wrong frequency ()presumably another spot). Is that not an issue with this change?

Tom NY4I

n4af, 2023-09-20:
it is/remains an issue.
I think beta2 reduces the updates substantially tho...

df5en, 2023-09-21:
4.126.beta2: I can confirm that.

n4af, 2023-09-21:
well, we are back to square one.
Locking the BM throws it out of sync with the buffer. 
Stopping the buffer means spots are thrown away.

4.37 introduced 'prevent refresh' -when focus was on the bm. Nice, but it causes the bm<>buffer to be out of sync as does writing buffer entries out in groups of two or three.

The only solution I am aware of is strong filtering of spots.
One feature that works well is ctrl-j parameter IE SWITCH - when set to 'true', only spots in trmaster or  the initial exchange file will go into the bandmap.

ny4i, 2023-09-21:
I do believe using critical sections around the code that  changes where the BM info is stored would help. I just had not made it that far. It has to be done carefully to make sure any exceptions don’t orphan a lock call (try…finally)TomPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771813-205-6388 (mobile/text)

n4af, 2023-09-24:
OK, the CQWWRTTY is underway and  a perfect chance to check for rapid updates/blinking.

Log into 'dx.k3lr.com:23' and set filter to 'set dx/filter skimmer':
spots flowing from ww skimmers are overwhelming !

Now, I  edit file 'cluster_commands.txt' (the tr4w cluster commands file)
I add a line so the AR COMMANDS reads:
#AR-CLUSTER COMMANDS:

set/dx/filter call=N4AF or (not skimmer and spottercont=NA) or ((skimmer and unique>2) and skimvalid)

and restart TR4W, connect to dx.k3lr.com:23 and then select COMMANDS from the dx cluster window.

One of the lines reads SET FILTER. When I click on it my command (and a lot of other ones previously set) show up.
I click on set/dx/filter call=N4AF or (not skimmer and spottercont=NA) or ((skimmer and unique>2) and skimvalid)
and suddenly spots roll out in a very organized way every few seconds.
Because the continent of NA is so large, the actual command I use is
set/dx/filter call=N4AF or (not skimmer and spottercont=NA) or ((skimmer and unique>2) and  not skimbusted and (spotterstate=[WV,MD,VA,PA,NC]))

-But point being, that, with proper filtering, the spots rate should not be a problem-

In the case of the k3lr cluster, Tim has excellent documentation at http://www.k3lr.com/w9zrx/AR-Cluster%20Filter%20Commands.pdf

ny4i, 2024-05-16:
The pull request is in a branch called fix-sub-mode-ssb. I forgot to create a new branch but merging the pull request should work. Again, this is a test release to be evaluated.

n4af, 2024-07-28:
I have created a[ beta release](https://tr4w.net/4.136/tr4w_setup_4_136.beta.exe) that needs testing.
the beta creates two telnet buffers. Until the second telnet buffer (SpotsBuffer) is written, no buffer processing occurs.
Once both buffers are written, normal telnet buffer processing occurs.