TR4W/TR4W #700: Rules menu items are grayed out

ny4i, 2023-10-07:
Under the tools menu, the rules items are grayed out. Not sure why but this should just be a simple case to open the URL and let Windows open it with the browser.

n4af, 2023-10-08:
From MainUnit.pas
if ContestsArray[Contest].WA7BNM = 0 then
    Windows.EnableMenuItem(tr4w_main_menu, menu_sk3bg_calendar,
MF_BYCOMMAND or
      MF_GRAYED);

n4af, 2023-10-08:
Ok.no calendar numbers means gray -- 

If  WA7BNM parm is edited with the menu number in the perpectual calendar it will work in 4.127.1:
https://www.contestcalendar.com/perpetualcal.php
so, wa7bnm for CQP lists as '140' in the perpectual cal, so VC needs to be modified as
' ({Name: 'CQP';                        }Email: nil;                      DF: 'california_cty';    WA7BNM:  140;. . . .'
This has been done in 4.127.1 for CQP as a template for what is required to make RULES work for WA7BNM calendar

n4af, 2023-10-08:
There shud be a menu for wa7bnm. it presently lives within sk3bg. Since it works, i left it for time being...

n4af, 2023-10-08:
https://tr4w.net/4.127/tr4w_setup_4_127.1.exe should work where the Wa7BNM # is entered. REF: CQP in VC.PAS

n4af, 2023-10-08:
line 3218 https://github.com/n4af/TR4W/blob/master/tr4w/src/VC.pas

ny4i, 2023-10-08:
Thanks. That works. It didn;t dawn on me that the entry might be missing. I thought I checked those previously to update them all. I will go through all th eocntests and check.

I did notice an extra space in the menu now though.

[image]

ny4i, 2023-10-08:
The bandmap does not appear to be working on this build. I went back to the .0 release and it works again.

n4af, 2023-10-08:
Seems to be working here.
i will have a look after lunch...

n4af, 2023-10-08:
yeah, there is a problem.

n4af, 2023-10-08:
Would you try d/l 4.127.1 again and verify the bm is now ok ??
Thanks, Howie

ny4i, 2023-10-08:
That works. Thanks

closed 2023-10-28 by n4af