TR4W/TR4W #702: Number of posted Q's to ContestOnlineScore is too high when there are deleted Q's

df5en, 2023-10-18:
TR4W: v.4.127.2
contest: WAG (eventually also in all other contests)
posting to: www.contestonlinescore.com

1. You have "n" logged QSOs and post it to www.contestonlinescore.com. You'll see your score with "n" QSOs on the webpage.
2. Now delete "m" QSOs (by selecting the attribute "Deleted" in the "View/Edit log" window) and then save it.
3. Press "Post now" and then refresh teh page www.contestonlinescore.com
4. Look at your QSO number on the webpage. You would expect now "n-m" QSOs, but there are still "n" QSOs.

This effect occurs only after deleting the QSOs in the View/Edit window. When you delete the last QSO with ALT-Y, the displayed number of QSOs on contestonlinescore is correct.

73's
Dieter

P.S. No hurry. In my opinion this is a typical low priority problem.

ny4i, 2023-10-18:
If you rescore after editing, is the count correct?Tom NY4I Principal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771813-205-6388 (

ny4i, 2023-10-18:
> Wow, thanks for extremely quick reply, Tom.
> A "Rescore" in the "Commands" menu doesn't change anything. The number of QSOs on contestonlinescore is still wrong.

Ok. Thanks for the bug report. We’ll take a look. TomPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771813-205-6388 (mobile/text)

ny4i, 2024-02-05:
I took a look at this issue. I had 2 15 meter test contacts in a log and the score message is as follows:

``` xml
04 Feb 2024 21:24:26.809 190962 [15304] debug TR4WDebugLog  - [MakePOSTRequestNew] POST / HTTP/1.1
Content-Type: application/x-www-form-urlencoded
Content-Length: 1878
Host: post.contestonlinescore.com
Connection: close

xml=<?xml version="1.0"?><dynamicresults><soft>TR4W v.4.131.1</soft><contest>WINTER FIELD DAY</contest><call>W4AFC</call><class ops="SINGLE-OP" mode="CW" power="HIGH" bands="ALL" transmitter="ONE"></class><breakdown><qso band="80" mode="DIG">19</qso><qso band="80" mode="ALL">19</qso><qso band="40" mode="DIG">11</qso><qso band="40" mode="ALL">11</qso><qso band="20" mode="DIG">4</qso><qso band="20" mode="PH">3</qso><qso band="20" mode="ALL">7</qso><qso band="15" mode="DIG">2</qso><qso band="15" mode="ALL">2</qso><qso band="10" mode="DIG">3</qso><qso band="10" mode="PH">3</qso><qso band="10" mode="ALL">6</qso><qso band="2" mode="DIG">1</qso><qso band="2" mode="ALL">1</qso><qso band="432" mode="DIG">1</qso><qso band="432" mode="ALL">1</qso><qso band="total" mode="DIG">41</qso><qso band="total" mode="PH">6</qso><qso band="total" mode="ALL">48</qso><mult band="80" mode="DIG" type="state">7</mult><mult band="80" mode="ALL" type="state">7</mult><mult band="40" mode="DIG" type="state">7</mult><mult band="40" mode="ALL" type="state">7</mult><mult band="20" mode="DIG" type="state">2</mult><mult band="20" mode="PH" type="state">3</mult><mult band="20" mode="ALL" type="state">5</mult><mult band="10" mode="DIG" type="state">3</mult><mult band="10" mode="PH" type="state">1</mult><mult band="10" mode="ALL" type="state">4</mult><mult band="2" mode="DIG" type="country">1</mult><mult band="2" mode="ALL" type="country">1</mult><mult band="432" mode="DIG" type="state">1</mult><mult band="432" mode="ALL" type="state">1</mult><mult band="total" mode="DIG" type="state">20</mult><mult band="total" mode="DIG" type="country">1</mult><mult band="total" mode="PH" type="state">4</mult><mult band="total" mode="ALL" type="state">24</mult><mult band="total" mode="ALL" type="country">1</mult></breakdown><score>792</score><timestamp>2024-02-05 02:24:26</timestamp></dynamicresults>
```

Then I used View / Edit log and checked the Deleted option to remove the contact. After I did that, this was the XML sent:

``` xml
04 Feb 2024 21:26:20.427 304580 [11664] debug TR4WDebugLog  - [MakePOSTRequestNew] POST / HTTP/1.1
Content-Type: application/x-www-form-urlencoded
Content-Length: 1878
Host: post.contestonlinescore.com
Connection: close

xml=<?xml version="1.0"?><dynamicresults><soft>TR4W v.4.131.1</soft><contest>WINTER FIELD DAY</contest><call>W4AFC</call><class ops="SINGLE-OP" mode="CW" power="HIGH" bands="ALL" transmitter="ONE"></class><breakdown><qso band="80" mode="DIG">19</qso><qso band="80" mode="ALL">19</qso><qso band="40" mode="DIG">11</qso><qso band="40" mode="ALL">11</qso><qso band="20" mode="DIG">4</qso><qso band="20" mode="PH">3</qso><qso band="20" mode="ALL">7</qso><qso band="15" mode="DIG">1</qso><qso band="15" mode="ALL">1</qso><qso band="10" mode="DIG">3</qso><qso band="10" mode="PH">3</qso><qso band="10" mode="ALL">6</qso><qso band="2" mode="DIG">1</qso><qso band="2" mode="ALL">1</qso><qso band="432" mode="DIG">1</qso><qso band="432" mode="ALL">1</qso><qso band="total" mode="DIG">40</qso><qso band="total" mode="PH">6</qso><qso band="total" mode="ALL">48</qso><mult band="80" mode="DIG" type="state">7</mult><mult band="80" mode="ALL" type="state">7</mult><mult band="40" mode="DIG" type="state">7</mult><mult band="40" mode="ALL" type="state">7</mult><mult band="20" mode="DIG" type="state">2</mult><mult band="20" mode="PH" type="state">3</mult><mult band="20" mode="ALL" type="state">5</mult><mult band="10" mode="DIG" type="state">3</mult><mult band="10" mode="PH" type="state">1</mult><mult band="10" mode="ALL" type="state">4</mult><mult band="2" mode="DIG" type="country">1</mult><mult band="2" mode="ALL" type="country">1</mult><mult band="432" mode="DIG" type="state">1</mult><mult band="432" mode="ALL" type="state">1</mult><mult band="total" mode="DIG" type="state">20</mult><mult band="total" mode="DIG" type="country">1</mult><mult band="total" mode="PH" type="state">4</mult><mult band="total" mode="ALL" type="state">24</mult><mult band="total" mode="ALL" type="country">1</mult></breakdown><score>774</score><timestamp>2024-02-05 02:26:20</timestamp></dynamicresults>
```

Notice that the 15 QSO 15m ALL is now 1 versus 2 in the prior xml string before the delete.

This seems to work. 

I will try it in the WAG contest you reported to see what happens.

ny4i, 2024-02-05:
Oh I think I see the issue.

After the delete the qso band total field is still the number before I deleted the contact. 

<qso band="total" mode="ALL">2</qso>

ny4i, 2024-02-05:
This looks better. 

```xml
xml=<?xml version="1.0"?><dynamicresults><soft>TR4W v.4.131.1</soft><contest>WAG</contest><call>NY4I</call><class ops="SINGLE-OP" mode="CW" power="HIGH" bands="ALL" transmitter="ONE"></class><breakdown><qso band="40" mode="CW">3</qso><qso band="total" mode="CW">3</qso><qso band="total" mode="ALL">3</qso><mult band="40" mode="CW" type="state">1</mult><mult band="40" mode="ALL" type="state">1</mult><mult band="total" mode="CW" type="state">1</mult><mult band="total" mode="ALL" type="state">1</mult></breakdown><score>9</score><timestamp>2024-02-05 04:10:46</timestamp></dynamicresults>
```

This is after deleting multiple contacts. I will finish testing it as well as make sure the same logic is added to the UDP message for WRTC if needed.

commit a5419481 (2024-02-05): Fix score posted to online scoreboard and UDP

The score sent to the online contest post was including the deleted contacts in the  total. This was incorrect. THe UDP sentfor score had the same issue. This was fixed as well. This closes #702.

Note I also moved some debug messages to the TRACE level to clean up the DEBUG output. This was mostly CWbyCAT related messages.