TR4W/TR4W #704: Entering 198 in the call window is changing to 40m (7.198) but was on 20

ny4i, 2023-10-28:
expected the radio to tune to 14198 since it was not enough to set the MHz. For some reason, the radio switched to 7.198 from 20m. K4 networking in case that matters.

n4af, 2023-10-28:
Another problem with adding 60M band.
Fix 4.127.5

n4af, 2023-10-28:
https://tr4w.net/4.127/tr4w_setup_4_127.5.exe

ny4i, 2023-10-28:
I’ll test it out. Thanks for fixing. BTW, I enter those as I run into them to address them later after contest. I’m not kicking it over the wall. :) But it doesn’t matter who fixes it. Just wanted to clarify. TomPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771813-205-6388 (mobile/text)

https://tr4w.net/4.127/tr4w_setup_4_127.5.exe

ny4i, 2023-10-29:
Confirmed working. Good fix. Thanks!

closed 2023-10-29 by ny4i