TR4W/TR4W #707: Add support to log QSOs directly to DXKeeper via TCP/IP

ny4i, 2023-11-26:
While there is a UDP gateway to log contacts, this requires running a gateway program. For casual DX chasing-type contesting, I would rather have a direct TCP/IP connection to DXKeeper to log contacts. One other benefit is to add a delay between the logged time and the actual time the contact is sent. This allows to edit the contact if necessary. That timer should be on a parameter (EXTERNAL LOG DELAY = 60 SECONDS).

This code should be put in its own unit to be the start of making the connection toother loggers as well. The option to log to an external program should be disabled by default.

The following information is how to send a contact to DXKeeper. Note that the DXLab uses a base port (defaults to 52000). The DXKeeper port listens on the base port + 1 so add EXTERNAL LOG BASE PORT = 52000 as a parameter.

> Interoperation via TCP
> Commander can be configured to respond to directives received in TCP messages, as described [here](https://www.dxlabsuite.com/commander/Help/NetworkService.htm#TCP_Network_Service). The format of these messages is documented [here](http://www.dxlabsuite.com/commander/Commander%20TCPIP%20Messages.pdf).
> 
> DXKeeper accepts a TCP message beginning with the externallog command directing it to log a QSO in a manner governed by multiple parameters:
> 
> <Parameters:N> - N is the total length of the ExternalLogADIF field and all specified option fields
> 
> <ExternalLogADIF:M>valid ADIF record of length M - the QSO to be logged
> 
> Options - a subset of the [General Options](https://www.dxlabsuite.com/dxkeeper/Help/Import.htm#Options) available for importing QSOs
> 
> <UploadeQSL:1>Y - if present, submit the logged QSO to eQSL
> 
> <UploadLoTW:1>Y - if present, submit the logged QSO to LoTW
> 
> <DeduceMissing:1>Y - if present, deduce items missing from the QSO to be logged from the entity, primary subdivision, and USAP databases
> 
> <QueryCallbook:1>Y - if present, query the [currently selected Callbook](https://www.dxlabsuite.com/dxkeeper/Help/Configuration.htm#Callbook%20Tab), DXCC database, and RDA database for missing items
> 
> <UpdateeQSL:1>Y - if present, query the eQSL database to determine if the callsign specified in the ADIF record is an authenticity-guaranteed member of eQSL, and if so set the QSO's [eQSL Member](https://www.dxlabsuite.com/dxkeeper/Help/Items.htm#eQSL.cc%20mem) item to A
> 
> <UpdateLoTW:1>Y - if present, query the eQSL database to determine if the callsign specified in the ADIF record has submitted QSOs to LoTW to LoTW within the [specified interval](https://www.dxlabsuite.com/dxkeeper/Help/QSLConfiguration.htm#Maximum%20age%20of%20most%20recent%20LotW%20upload), and if so set the QSO's [LotW Member](https://www.dxlabsuite.com/dxkeeper/Help/Items.htm#LotW%20mem) item to Y
> 
> <CheckOverrides:1>Y - if present, query the [Override list](https://www.dxlabsuite.com/dxview/Help/Configuration.htm#Entity%20Overides) for the presence of the callsign specified in the QSO to be logged; if found, update the logged QSO with the information specified in the override
> 
> <UploadClubLog:1>Y - if present, submit the logged QSO to Club Log
> 
> <UploadQRZ:1>Y - if present, submit the logged QSO to QRZ
> 
> For example, receipt of the message
> 
> <command:11>externallog<parameters:341><ExternalLogADIF:187><CALL:4>P5DX <RST_SENT:3>599 <RST_RCVD:3>599 <FREQ:6>14.004 <BAND:3>20M <MODE:2>CW <QSO_DATE:8>20220411 <TIME_ON:6>072800 <STATION_CALLSIGN:5>AA6YQ <TX_PWR:4>1000 <GRIDSQUARE:4>PM29 <EOR><UploadeQSL:1>Y<UploadLoTW:1>Y<UploadClubLog:1>Y<DeduceMissing:1>Y<QueryCallbook:1>Y<UpdateeQSL:1>Y<UpdateLoTW:1>Y<CheckOverrides:1>Y
> 
> would result in logging a QSO with P5DX on 140004 kHz in CW on 2022-04-11 at 0728Z in Grid PM29 with 1000 watts; the override list would be checked for the presence of P5DX, missing items would be deduced from available databases, the currently-selected callbook would be queried for additional information, the logged QSO's eQSL and LoTW membership items would be set to Y, and the logged QSO would be submitted to Club Log, eQSL, and LoTW.
> ```

closed 2024-01-21 by n4af

ny4i, 2024-01-21:
Thanks. If you have not, can you put an installer for this under the latest directory so I can test?ThanksTomPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771813-205-6388 (mobile/text)

Closed #707 as completed via #714.

n4af, 2024-01-21:
Hi Tom
Shud b Url  on end of issue707 ?

PR #714 (2024-01-21): Issue 707 external logger
This closes #707 and implements a change to the LogRadio UDP as well as the LogContact UDP. This also updated the error message in the language constant files to indicate that possible Winter Field Day classes now include M.