TR4W/TR4W #708: For Croatian contest file 'croat.dom' is missing

df5en, 2023-12-11:
For the Croatian contest TR4W stops starting because the file 'croat.dom' is missing.

Page 3 of https://www.hamradio.hr/wp-content/uploads/2023/09/9A_DX_prevod_na_hrvatski.pdf describes the Croatian districts. Now with the file 'croat.dom' TR4W works.

bj=BJ,^Bjelovar
bm=BM,^Beli Manastir 
ck=CK,^Cakovec
da=DA,^Daruvar
de=DE,^Delnice
du=DU,^Dubrovnik
dj=DJ,^Dakovo
gs=GS,^Gospic
im=IM,^Imotski
ka=KA,^Karlovac
kc=KC,^Koprivnica
kr=KR,^Krapina
kt=KT,^Kutina
kz=KZ,^Krizevci
ma=MA,^Makarska
na=NA,^Nasice
ng=NG,^Nova Gradiska
og=OG,^Ogulin
os=OS,^Osijek
pu=PU,^Pula
pz=PZ,^Pozega
ri=RI,^Rijega
sb=SB,^Slav'ski Brod
sk=SK,^Sisak
sl=SL,^Slatina
st=ST,^Split
si=SI,^Sibenik
vk=VK,^Vinkovci
vt=VT,^Virovitica
vu=VU,^Vukovar
vz=VZ,^Varazdin
zd=ZD,^Zadar
zg=ZG,^Zagreb
zu=ZU,^Zupanja

(I shortened the text of district 'SB' slightly. Also I did change special Croatian chars to ASCII chars.)

n4af, 2023-12-12:
fix[ 4.129.1](https://tr4w.net/4.129/tr4w_setup_4_129.1.exe)

df5en, 2023-12-12:
Test run of 4.129.1 works. Just in time for this contest on Saturday. Thanks. 

Vy 73
Dieter

n4af, 2023-12-12:
Thanks for testing Dieter...

closed 2023-12-12 by n4af