TR4W/TR4W #710: Feedback CROATION Contest: ITU-Zone / 9A-District in GUI and "MY QTH=" in CROATIAN.CFG

df5en, 2023-12-18:
v.4.129.1

In the CROATIAN Contest, there is no obligation to enter your own ITU zone (or, as a Croatian station, the two-letter district) immediately after selecting the contest.

Since no "MY QTH=xx" is then written into the CFG file, the later Cabrillo generation must fail (serial numbers instead of static "599 ITU-ZONE" or "599 DISTRICT"). The ADIF is also messed up because TR4W writes an error text to the STX_STRING.

Suggestion: Same procedure as for RAEM contest selection

1. When initializing the CROATIAN Contest for the first time, the user must enter their own ITU zone. Or, as a 9A station, he must enter his two-letter district.

2. Then write this information to the CROATIAN.CFG file as "MY QTH=xx". 

3. That's it. Cabrillo and ADIF are then created correctly.

I verified this in a dry run after I had to change 585 QSOs manually with Notepad in the real contest Cabrillo. It was just my mistake. I should have checked Cabrillo and ADIF **BEFORE** the contest. ;-) 

Vy 73
Dieter
DF5EN@DF0SX

n4af, 2023-12-18:
4.129.2 (issue 709) addresses that by looking if the entry station is in '9A'.
However there are still problems addressed in #709..
At this time, I am not sure I want to accommodate  <column requirements for the cabrillio> -- IMO,
that should not occur- parsing should be based on white space delimitation of fields.
Response from sponsors who refuse to parse by whitespace:
------
Log columns (QSO:) must be in the following format:

QSO:  freq mo date       time per call        rst ex1 ex2 call         rst ex1 ex2
QSO: ***** ** yyyy-mm-dd nnnn *** ************* nnn nnn aa *********** nnn nnn aa
QSO:  3501 CW 2006-10-14 1403   9A2HM         599 005    9A3MA         599 008 SB
000000000111111111122222222223333333333444444444455555555556666666666
123456789012345678901234567890123456789012345678901234567890123456789

df5en, 2023-12-18:
v.4.129.2 seems to work and does not have the problem of v.4.129.1.

In my opinion, this point can therefore be closed.

closed 2023-12-18 by n4af