TR4W/TR4W #712: DARC 10m contest - rules change

df5en, 2024-01-21:
Rule change in the DARC 10-meter contest: Mixed mode only. No change in CW-only and no change in SSB-only

The next DARC 10 meter contest is in January 2025. So don't rush.

English: https://www.darc.de/der-club/referate/conteste/darc-10m-contest/rules/
German (the changes in red font): https://www.darc.de/der-club/referate/conteste/darc-10m-contest/rules/

I understand it like this:

Old, until 2023: After you had a QSO with a station in CW, a QSO afterwards with the same station in SSB counted neither points nor multis.

New, from 2024: After you had a QSO with a station in CW, a QSO with the same station in SSB counts both points and multis (if this WAE/DXCC/DOK multi is new in SSB and not already a third station has assigned this).

The same applies in reverse if you first have a QSO in SSB and then in CW.

73
Dieter

n4af, 2024-02-06:
fix [4.131.5](https://www.tr4w.net/4.131/tr4w_setup_4_131.5.exe)

closed 2024-02-06 by n4af

df5en, 2024-02-06:
Yes, I tested version 4.131.5 in a small dry run with some call signs and DOKs in CW and SSB. Points and multi calculations look correct.

Thanks Howie.

Vy 73
Dieter

df5en, 2024-02-06:
Oops. Sorry for re-opening. Can be closed.