TR4W/TR4W #713: RAEM contest, error text in STX_STRING after ADIF generation

df5en, 2024-01-21:
v.4.129.2 and v.4.130.0

Here is feedback from the RAEM Contest follow-up. Unexpectedly, there was a small problem with the ADIF file after the contest. I had never seen this effect in any other contests of TR4W. The STX_STRING contained the text **Error generating my exchange**, even though it should contain geographical coordinates. In my case STX_STRING should be a **51N07O**.

This is how I declared the parameter in the CFG file:
`MY QTH=51N07O`

Here is the first data line of the ADIF. All other lines look the same.

Error text in ADIF output:
`<CALL:4>YT3D <BAND:3>80m <QSO_DATE:8>20231224 <TIME_ON:6>000033 <TIME_OFF:6>000033 <RST_SENT:3>599 <RST_RCVD:3>599 <CONTEST_ID:4>RAEM <MODE:2>CW <FREQ:7>3.51512<SRX_STRING:7>1 4N19O <STX_STRING:28>Error generating my exchange <QTH:6>45N19O <SRX:5>00001<STX:5>00001 <OPERATOR:5>DF5EN <EOR>`

Correct, post-processed:
`<CALL:4>YT3D <BAND:3>80m <QSO_DATE:8>20231224 <TIME_ON:6>000033 <TIME_OFF:6>000033 <RST_SENT:3>599 <RST_RCVD:3>599 <CONTEST_ID:4>RAEM <MODE:2>CW <FREQ:7>3.51512<SRX_STRING:7>1 4N19O <STX_STRING:6>51N07O <QTH:6>45N19O <SRX:5>00001<STX:5>00001 <OPERATOR:5>DF5EN <EOR>`

It wasn't a real problem as everything could be fixed with a Search&Replace in the editor. Luckily the value to be replaced was just a constant. Nevertheless, the feedback here is that there is a small problem with ADIF generation (only) in the RAEM contest. Everything else worked perfectly and stably throughout the entire contest.

Vy 73
Dieter

ny4i, 2024-02-06:
I found the issue but STX_STRING is the whole exchange so it would be the serial number and the coordinates. In my test case, 

<CALL:5>DF5EN <BAND:3>80m <QSO_DATE:8>20240206 <TIME_ON:6>005033 <TIME_OFF:6>005033 <RST_SENT:3>599 <RST_RCVD:3>599 <CONTEST_ID:4>RAEM <MODE:2>CW <SRX_STRING:8>1 57N85O <STX_STRING:6>51N07O <QTH:6>57N85O <SRX:5>00001<STX:5>00001 <OPERATOR:4>NY4I <EOR>

Does that work for you?

df5en, 2024-02-06:
> Does that work for you?

Yes absolutely. Checked it with my log file from December 24, 2023 in version 4.131.5. There are no differences to my modified ADI file. Except for the header, both files are absolutely identical.

> I found the issue but STX_STRING is the whole exchange so it would be the serial number and the coordinates.

As it is now in v.4.131.5 I am happy because QRZ.com accepts the ADI file.

In my opinion #713 can be closed. Thanks Tom.

Vy 73
Dieter