TR4W/TR4W #740: The BAND MAP CALL WINDOW feature does not seem to work

ny4i, 2024-03-03:
I am in search an pounce mode. I switch the radio manually to a frequency where a known station is in the BANDMAP. It used to work where the call in the bandmap would automatically populate the call window when it got within a number of Hz. That does not appear to work. I checked the code but I have not yet figured it out since it does really odd things like compare the spot frequency against tCQAsInteger.

I did see some comments with N4AF worked int he code soi Howie may understand it more to fix it.

ny4i, 2024-03-03:
Scratch that. This works OK when tuning the VFO. Where it does not work is when I set the rig to a specific frequency with another program so I will check that out.

ny4i, 2024-03-03:
I may not have been in S&P mode. But still checking. I am curious what tCQAsInteger is used for...

n4af, 2024-03-04:
FWIW,  VC defines it as const  'CQ/ '

ny4i, 2026-11-04:
Closing as not reproducible: the reporter found it works when the VFO is tuned and only failed when another program set the frequency. The cross-platform TR4W (TR4W/TR4W-D12) has a bench check for the band map filling the call window; any new report goes there.

closed 2026-11-04 by ny4i