TR4W/TR4W #745: Contest name change from "TESLA" to "HF-TESLA"?

df5en, 2024-03-13:
I believe the name for the TESLA contest was changed from "TESLA" to "HF-TESLA". At least at contest.run the TESLA contest now appears under HF-TESLA: http://contest.run/#/help/develop

I took part in this contest and my call didn't appear on ContestOnlineScore or contest.run.

The same applies to the Cabrillo file. I think that here too the contest name should be HF-TESLA instead of TESLA. (I saw other logs that had an HF-TESLA there)

Is my guess correct?

ny4i, 2024-03-13:
The Cabrillo name is TESLA-HF. I will update when I review the other data. Thanks for sending it.

https://www.contestcalendar.com/contestdetails.php?ref=566

ny4i, 2024-03-13:
Note to self…Also add the code 566 to the WA7BNM: 0000; field which is now 0000.

justzh, 2024-04-04:
No