TR4W/TR4W #752: Wrong Zone logged in CQ WW

ny4i, 2024-04-02:
While doing some testing, I noticed that if I have CQ-WW-CW running, when I log a US call of W4RD with a zone of 3 (like I used to do when I lived in Utah as NY4I in zone 3 instead of zone 5), the RxData.QTH.Zone is set to zone 5. 

I would think if the exchange was a zone, this should override the zone from any lookup. I have not yet checked but it would appear this is being overridden by the zone lookup instead of what was actually entered.

I'm looking for confirmation to make sure I am not doing something else wrong.

n4af, 2024-04-02:
AFAIK, zone is pulled from cty.dat.
Note: only exceptions to call area/zone are recorded. So something like
4u1wb (DC) is recorded in cty.dat
 =4U1WB(5)[8] for CQ/IARU zones.

ny4i, 2024-04-02:
Right but if I work a 4 station on Utah (zone 3) and I log 3 which they give me, then I would think that should be put in the RxData.QTH.Zone instead of what it gets from cty.dat.

If this is letting cty.dat override the zone entered int he exchange, I would call that a bug.

Tom

n4af, 2024-04-02:
agreed. but i cannot seem to duplicate.
 if i  log ny4i as '"3" it logs as "3" ??
QSO:  1800 CW 2024-04-02 1801 N4AF    599 05     W4PD           599 03

QSO:  1800 CW 2024-04-02 1802 N4AF    599 05     W3AU           599 06

QSO:  1800 CW 2024-04-02 1803 N4AF    599 05     NY4I           599 03

ny4i, 2024-04-02:
I agree. I also checked the ADIF file and it shows up as logged.

<CALL:4>NY4E <BAND:3>15m <QSO_DATE:8>20240402 <TIME_ON:6>165507 <TIME_OFF:6>165507 <RST_SENT:3>599 <RST_RCVD:3>599 <CONTEST_ID:8>CQ-WW-CW <MODE:2>CW <FREQ:6>21.074<SRX_STRING:1>3 <STX_STRING:6>599 05 <CQZ:2>03<OPERATOR:4>NY4I <EOR>

So the issue is there is most likely code looking at the exchange but the RXData.QTH.Zone is set to 5. Should I think of the QTH.Zone as the info derived from the cty.dat location and not the info from the exchange?

That is OK as I can grab the value from the exchange if I need to do so.

Tom

ny4i, 2024-04-02:
I found it. I was using RXData.QTH.Zone instead of RxData.Zone. TomPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771813-205-6388 (mobile/text)

closed 2024-04-02 by ny4i