TR4W/TR4W #758: Add date-aware contest selection to contest drop-down

ny4i, 2024-04-26:
Show the contests active in the current weekend first in the contest drop-down. Have a section up top with the contests. Use the contest calendar CSV file to find the contests for "this weekend"

Idea originally from POLO developer,

sd, 2024-04-26:
That's an awesome idea 😁

n4af, 2024-04-27:
Does 'value added' exceed the effort to duplicate what already is extant ?

My preference is a single data source- that being[ WA7BNM weekly, monthly, yearly, and perpetual calendar](https://www.contestcalendar.com/weeklycont.php#16496).
Additionally, larger clubs maintain a list of contests qualifying for in-club awards.
Also, keeping a calendar up to date can be 'challenging'.

ny4i, 2024-04-27:
WA7BNM’s data was going to be my source. It’s not terribly hard to implement and the  usability factor is very high.While we have a real-time source of upcoming contests, I was also thinking to put a data code in out contests array with the month and “which weekend” so it was a static list. Then we just include the contests for this week. Additionally I always wanted to learn how to write a custom drop-down in Win32. Once I see if this is possible, I was going to get more feedback from the user base as well.

closed 2026-11-04 by ny4i