TR4W/TR4W #759: version.pas still has 4.133.2

ny4i, 2024-04-28:
I noticed the website shows the may 2024 version as 4.134 but version.pas still shows 4.133.1 (as does the program title window). The link behind  the website link still shows 4.133 as well.

n4af, 2024-04-28:
Ok, thanks for the head up. I will regenerate it.

n4af, 2024-06-05:
arg. i did update but somehow the old took...

ny4i, 2024-06-05:
Do you always do a full get on master before you build? That would ensure you are getting the latest files before every build. This command does a force to pull everything and overwrite anything local.

git reset --hard origin/master

One thing I find useful for building especially in a multi-developer environment is to setup a VM with Windows and Delphi and use that exclusively for building. That also ensures I do not accidentally install a local unit into my development environment and then forget to check it in so the next developer cannot compile. Just a thought...

ny4i, 2024-06-05:
I was just closing this prior as I thought we were ok. The version I just downloaded from the website was 4.136.0.4. That's the latest, right?

n4af, 2024-06-05:
no. i have lost my mind and cannot move good 4.136.0 up to tr4w.net and
download w/o it being 136.0.4
no idea how it is ending up that way...

closed 2024-07-28 by n4af