TR4W/TR4W #760: Tune Spot using RadioTwo

ny4i, 2024-04-28:
This is strange. Between the line, 

 TuneRadioToSpot(TempSpot, RadioOne); in TelnetWndDlgProc in uTelnet

and procedure SetRadioFreq(Radio: RadioType; Freq: longint; Mode: ModeType; VFO: char); in LogRadio, the radio object was passed as RadioOne but appears in the debugger as RadioTwo. 

Were there any changes recently that could have affected the active radio pointer?

n4af, 2024-04-28:
sigh, yeah i will take a look.

73, Howie

ny4i, 2024-04-28:
This works on 4.132.3, 4.133.1 but does NOT work as of 4.133.2

Hence one could conclude one of the changes in 4.133.2 broke it. That would be this change according to the release notes:

4.133.02 #757 SO2R fails to switch band & freq display (N4AF)

If you are doing something with the activeRadioObj you may need to lock that code if the Radio passes to SetRadioFreq is a pointer. Or dereference the pointer and p[ass the actual object (but I have not examined it enough to know if that is the issue).

n4af, 2024-04-28:
Hi Tom -

Could you try 4.134.1 <https://tr4w.net/4.134/tr4w_setup_4_134.1.exe>to
verify all is ok ??

Tks, Howie

ny4i, 2024-04-29:
Yes - that works Howie. Thanks

closed 2024-07-28 by n4af