TR4W/TR4W #763: Seems to be an issue with the IN7QPNE DOM file

ny4i, 2024-05-05:
I just noticed that the  DOM file in the build has the NE countries inverted. Instead of VTWNH, the DOM has WHNVT. That is incorrect. I will check to see what the DOM checked into GitHub has.

ny4i, 2024-05-05:
IN7QPNE.dom has not been updated in 3 years so we may have missed an update. I thought I saw some traffic on the reflector about this not long ago. I will check there if someone has a new file.

ny4i, 2024-05-05:
It looks like the latest file did not make it into MASTER on GitHub. It looks OK in the release. I was running it from Delphi.

I still after all these years am not quite sure how we can end up with a file in the build that is not committed into MASTER. Do you know how that happens in the case of a DOM file. After a build, I would expect everything in the build to be the same as what is downloaded from MASTER.

Tom

n4af, 2024-05-05:
the github of in7qpne_cty indicates the update was dropped. i will put out
a new refresh.

tks, Howie

n4af, 2024-05-05:
SIGH. Life is tough.

the file I was sent is not a lot better than the old version.

neqp county abbreviations https://neqp.org/neqp-county-abbreviations/  is
not ascii friendly to developers (as usual) -
so I will need to take some time to callup
a script that converts the insanity to 'county-abbrtev  (space) county
name'  in two columns.

ny4i, 2024-06-05:
The counties are correct in the latest download so it looks like the right DOM made it into the release. I tested and the NE counties have the state first. They are in lower case for some reason but they are still processed properly.

Closing.

closed 2024-06-05 by ny4i