TR4W/TR4W #773: Add native support for the Arizona QSO Party

ny4i, 2024-07-12:
A user  (Ross NS7F) request to add support via config files. See here: https://github.com/n4af/TR4W/issues/362

But add native support to the program.

ny4i, 2024-07-12:
A few scoring notes...

BONUS POINTS:
Receive a one-time bonus of 100 points for a QSO with K7A on any band or mode during the contest.

Total Score = (QSO POINTS x MULTIPLIERS) + BONUS POINTS.

MULTIPLIERS
Arizona stations
Multipliers are the 50 US states, the 13 Canadian provinces and territories, and DXCC countries. Multipliers count again for each mode. Total multipliers = (50 + 13 + DXCC) x 2.

Stations outside of Arizona
Multipliers are the 15 Arizona counties. Multipliers count again for each band and mode. Total possible multipliers = 15 x 6 x 2 = 180.

QSO POINTS: CW=2; Phone=1

ny4i, 2024-07-12:
From the other closed issue:

CONTEST TITLE = ARIZONA QSO PARTY
EXCHANGE RECEIVED = RST DOMESTIC QTH
DOMESTIC FILENAME = AZ.DOM
CONTEST NAME = ARIZONA QSO PARTY
DOMESTIC MULTIPLIER = DOMESTIC FILE
QSO POINT METHOD = ONE PHONE TWO CW
MULT BY BAND = TRUE
MULT BY MODE = TRUE
MULTIPLE BANDS = TRUE
MULTIPLE MODES = TRUE

n4af, 2024-07-23:
fix [4.136.2](https://tr4w.net/4.136/tr4w_setup_4_136.02.exe)

ny4i, 2024-08-14:
I recall a post about setting the Cabrillo name. I see the program has the array set to ArizonaQP but in the Contest Calendar on WA7BNM is is listed as AZ-QSO-PARTY (or an alias of AZQP).

Was there some other info from the sponsor on the actual name? Bruce's website could have an error of course especially if the sponsor's made a change.

Tom NY4I

AZ-QSO-PARTY

ny4i, 2024-08-15:
From the sponsor: 

Hi Tom,

AZ-QSO-PARTY is the Cabrillo name, and AZQP is the Cabrillo name alias. Either will work.

73,
Bill Clark, K6WSC
Arizona QSO Party Chairman

n4af, 2024-08-15:
Looks like the source is already updated.
Push to 4.137.1

n4af, 2024-08-15:
ok. i think git has the right cab name now ?

closed 2024-08-15 by ny4i