TR4W/TR4W #775: Add support for labre contest

n4af, 2024-07-20:
Add support for Brazilian LABRE per rules at URL:
https://mail.google.com/mail/u/0/?shva=1#inbox/FMfcgzQVxlGzrJqwXqZzkgKfQGSGbWxG?projector=1&messagePartId=0.1

ny4i, 2024-07-20:
Hi Howie - Please let me know when you have committed the changes to MASTER. I have a few changes to make so I wanted to pull the latest code. Thanks.

n4af, 2024-07-20:
Hi Tom -

I have only built the brazil.dom- tho suspect 90% of the remainder can be
'reused' from other contests.

73, Howie

ny4i, 2024-07-20:
Right but wouldn’t there be changes to VC for the ContestsArray?TomPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771813-205-6388 (mobile/text)

n4af, 2024-07-20:
Yep - i said 90%, things like VC arrays and logstuff have to have additions.
+
No AE for RSTDOMESTICorContinent

73, Howie

n4af, 2024-07-20:
this seems coded up EXCEPT for LABREQSOPointMethod  points +( testing).
More work tomorrow.

n4af, 2024-07-20:
correction on rules URL:
https://www.labre.org.br/contest/en/regulamento/

n4af, 2024-07-21:
4.136.1 is out for testing

n4af, 2024-07-23:
no feedback.
close to[ 4.136.2](https://tr4w.net/4.136/tr4w_setup_4_136.02.exe)

closed 2024-07-23 by n4af

ok1rr, 2024-07-23:
4.136.1 works! TNX so much, also the scoring is as expected!
Tested in the final part of the contest. The only rather cosmetic quirk 
is that their robot needs the contest name LABRE-DX (with the hyphen).

Great work, we have another contest working!

73,
Martin, OK1RR

Dne 21. 07. 24 v 13:12 Howie Hoyt napsal(a):

n4af, 2024-07-23:
Thanks for the feedback.
I have put a new 4.136.2 up that generates LABRE-DX in the cabrillo.

ok1rr, 2024-07-24:
> Martin,
>
> LABRE-DX should be the Cabrillo name?
>
> That is easy enough to update in the ContestArray in VC so please
> confirm if you are referring to the Cabrillo name.
>
> Thanks,
>
> Tom NY4I

Yes, exactly LABRE-DX

TNX, 73

m.

Dne 23. 07. 24 v 20:00 Tom Schaefer napsal(a):