TR4W/TR4W #788: NextBandMap still displays deleted spot

n4af, 2024-10-31:
4.138.4 refreshes flist before running the spotslist to  ensure the spotslist is updated with any deletes.

ny4i, 2024-10-31:
HowieI’m surprised the list isn’t updated after any delete. Wouldn’t that be something we always would want?TomPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771813-205-6388 (mobile/text)

n4af, 2024-10-31:
> 2024, at 8:51 AM, Howie Hoyt ***@***.***&gt; wrote: <br> 4.138.4 refreshes

Hi Tom -
Definitely. Not really a problem with delete.

Problem being NextBandMap was not refreshing the spotslist.

ny4i, 2026-11-04:
Already fixed in the cross-platform TR4W (TR4W/TR4W-D12, FreePascal/Lazarus): the band map NextBandMap rebuilds the spot list first; a check is in the bench queue. Closing here; any new report goes to TR4W-D12.

closed 2026-11-04 by ny4i