TR4W/TR4W #790: when tuning across call that is in  initial exch, when exch is loaded, the cursor should return to call window

n4af, 2024-11-23:
This a real hindrance. Tuning the band, when a call has a corresponding known exch and the exch is loaded into the exchange window, the cursor should return to the Call Window.

ny4i, 2024-11-24:
I looked for this behavior while operating last night. I saw the call pop into the window but I did not see the cursor move. I did not try to type anything so I will give that a go.

ny4i, 2024-11-24:
I wonder if what I just saw is related. When I have Initial Ex set to Zone, when I click on a spot in the bandmap while S&P, the first enter I hit logs the contact rather than send my call. 

But if I just enter the call, it works ok. 

It seems the program needs to know when the info is populated from the bandmap, to perhaps not populate the Initial Ex until the first return is hit on the call.

Put another way, clicking on the bandmap should just populate the call window and not the Exchange window with the Initial Ex.

n4af, 2024-11-24:
Hi Tom -
somewhat bizarre. The problem occurs NOT with the prefill, but when tuning
away. The spot clears but the cursor stays in the exch window.
Hope this is not because I am set for so2r.  I spent some time on it, but
no luck so far.

I may have introduced another bug related to space bar on a s&p entry ?
Thinking it used to call the station.

ny4i, 2024-11-25:
Should I make this issue where with Initial Exchange set, the first enter in the call window logs the contact? Or is that related to a prior change?

ny4i, 2024-11-25:
> Hi Tom -
>
> I don't think so. Users know to do a ctrl-enter if they want to get
> the exch w/o logging....

That’s not exactly what is happening.

This is different than prior behavior.

Do this test….Set Initial Exchange to None. 
Get some DX spots and run a test log for CQ WW CW. 
Go into S&P mode. 
Click on a spot and it fills in the call. Hit enter and it sends your call as expected then moves to the exchange field for you to enter the exchange.

Now set Initial Exchange to Zone.

Click on a spot and notice that the callsign populated the call window and the zone goes into the exchange window. Now hit Enter.

Rather than sending your call as I would expect, it logs the contact. That is not correct. It is also not how it works before (but I cannot tell you how far before).

I have not tried what happens when the log is say FIeld Day and it populated the exchange with the prior exchange for the second time you work a station but I will check that…

Regards,

Tom NY4I

n4af, 2024-11-25:
Hi Tom -
admittedly this changed with auto init-exch fill.

AFAIK, if the call AND exch are valid, you want  enter to log it ?

I have exch F1 key set to MY CALL. After two tries I move the dial which
clears the  fields.

OTH, if exch is empty, ENTER should dump your call.
Now, users who like the old system need to set the initial exchange to NONE.

Does that make sense?

ny4i, 2024-11-25:
One of the strongest features of TR was what is now called ESM by other loggers. What is supposed to happen is if the initial call has not been sent, just my call is sent. Operating without Initial Ex makes it not very useful. This is an easy fix though. When you click on a spot, just don’t fill in the exchange. Let hitting enter on the call do that just as if I had typed in the call without a spot. That is what changed. Is auto init-exch fill an option I can turnoff to restore the behavior? I wasn’t aware of that as a new command. TomPrincipal Solutions ArchitectBetter Software Solutions, Inc. 727-437-2771813-205-6388 (mobile/text)

n4af, 2024-11-25:
> What problem did Auto address? If it is useful, no reason to break it.
>
> Is it worth having an an option?
>
> Tom

probably not. i made a change for Peter and thought it would be a nice
addition. For all I know there are other problems.

I already pulled it out.

n4af, 2024-11-25:
This feature probably causes more trouble than helps. 
In 4.139.1 will remove the autofill for zones.
Auto fill is still operative via an InitialExchange file.

n4af, 2024-11-25:
4.139.1 should remove

closed 2024-11-25 by n4af

ny4i, 2024-11-26:
It works again as expected in 4.139.1 except for the regression of entering a frequency not clearing. See #794