TR4W/TR4W #800: In CW Speed Sync, program is not following the radio for a K4

ny4i, 2024-12-14:
With a networked K4 (and serial), when the CW speed is changed on the radio, the program is not updated. I do not recall if this is supposed to work. I thought it was sync'ed both ways so if the radio is changed, the program speed is changed but could be wrong.

ny4i, 2024-12-18:
This may not be done. I see some issues now with the speed display on the program following the radio properly. More testing but let me look at it for a few days before we decide to back this out.

n4af, 2024-12-18:
Well, for my K3's this works (program to keyer). 
I am not  aware of it ever working radio manual speed adj to program speed.
eg, one way from program to radio speed but manual adj does not change my program speed setting.

PR #804 (2024-12-16): Fixed CW speed following Kenwoods and Elecraft
Icom, Yaesu, Flex and HamLIb have to be done yet.

This should be OK to add. It fixes  #800 but still needs to be added for the radios above.
closed 2026-06-08 by ny4i