TR4W/TR4W #813: Too short field for call sign in cabrillo export

vksadek, 2025-02-23:
Since 4.106.7 there is only 8 characters for my call sign. 
In case of call sign with 8 characters, there is no space between call sign and report. 
Is there any reason for it?

n4af, 2025-02-23:
In a universe of infinite variables, my guess is this was an unintended consequence from another fix.

vksadek, 2025-02-24:
I've added minimal-side-effect fix. 
https://github.com/n4af/TR4W/pull/814

n4af, 2025-02-24:
Thanks for coding. [Add to 4.141.1](https://tr4w.net/4.141/tr4w_setup_4_141.1.exe)

closed 2025-02-24 by n4af

n4af, 2025-02-25:
> The Cabrillo spec doesn&#39;t mention a limit on the call field. Would it
> not make sense for us to just set the field to our max callsign field? Then
> the code in question would become something like this: <br> <br>
> wsprintf(CABRILLO_FIRST_PART, &#39;QSO: %5s %s %d-%02d-%02d %02d%02d
> %-$SIZE(ceCall)s ‘); <br> <br> This part, $SIZE(ceCall) is pseudo code as I
> always have to lookup the exact Delphi syntax to build a variable format
> string. <br> <br> Moving this away from the assembler wsprintf stuff would
> be good. I try to change those whenever I make any changes to an area. <br>
> <br> Tom NY4I <br> <br> &gt; On Feb 24, 2025, at 7:10 AM, Howie Hoyt

Possibly more eloquent.
The immediate problem was handling 10 char calls.
ka6bim/kl7 would be an example.